makerbase-mks/MKS-WIFI #5: STA Disconnected (Robin nano / MKS WiFi / Sapphire pro)

Sergey-Afonin, 2020-05-25:
I put DIY module based on ESP12F to Robin nano and upgraded firmware.
I can  see IP address of printer on screen (obtained via DHCP) and Wifi name, but status is STA Disconnected. Meanwhile I can ping it from the network and can communicate with it't configuration screen via web page. I know one more person with the same issue. We used MSK formware and TwoTrees fork. Both versions do the same and have the same behevoiur and messages.
What can be wrong here?
[image]
[image]
[image]

Sergey-Afonin, 2020-06-07:
Thank you! I'll do own job to resolve.

closed 2020-06-07 by Sergey-Afonin

Onegin45, 2020-06-13:
Сергей, здравствуйте. Имею подобную проблему, с той лишь разницей, что принтер ip и символ облака на своем экране не показывает показывает. Вебсайт с зеленым фоном тоже доступен. Слайсер принтера не видит. Вы как-то победили эту проблему?

Sergey-Afonin, 2020-07-12:
> Сергей, здравствуйте. Имею подобную проблему, с той лишь разницей, что принтер ip и символ облака на своем экране не показывает. Вебсайт с зеленым фоном тоже доступен. Слайсер принтера не видит. Вы как-то победили эту проблему?

Я купил готовый модуль вместо самодельного. Понимаю, что здесь описано не всё. Только прошивки не хватает. Нужно как-то еще настроить модуль.

Onegin45, 2022-09-17:
Извините, что поздно. Если модуль покупной, то все должно работать из
коробки. Если вебсайт доступен, то модуль исправен и видит сеть. Смотрите в
настройках слайсера. У меня принтер в Куре периодически теряется - помогает
удалить\настроить заново подключение по Wifi