makerbase-mks/MKS-WIFI #9: Passphrase length limited to 18 chars

mikroled, 2020-09-15:
In germany the common routers Fritz!Box with a predifed WiFi Key with a legth of 20 characters (20 numbers).
The gui of the Ronbin Nano/MKS-Wifi only excepts up to 18 characters.

Cloud you please extend the size of the accepted passphrase in the gui up to minimal 20 characters?

makerbase-mks, 2020-09-16:
MKS has limit display to 18 characters, but you can use it normal.

mikroled, 2020-09-17:
The gui only accepts 18 characters and is not useable for configuration WiFi settings where passwords with more characters are used.
You need to configure the WiFi settings in the board configuration file robin_nano35_cfg.txt to get it working.

khaoskosmos, 2020-11-21:
This is a problem in the USA as well, many default passwords are over 17 characters... big problem. The config text file only works on non-Marlin firmware (ie. OLD), correct?