makerbase-mks/MKS-WIFI #11: MKS Wifi with ESP3D Firmware do not communicate with MKS Robin Nano

easycorexy, 2020-11-02:
Hello,

i have installed the ESP3D Firmware on the MKS Wifi Module. The firmware works fine and the WebUi also is also ok.

I have the following problem: **The Wifi-module doesn't communicate with the motherboard.**

I have MKS_Robin_Nano_V1.2 Board with Mks-Robin-Nano-Marlin2.0-Firmware on it.

Can you help telling me what do have to do to make this work?

Thanks in advance

mks-viva, 2020-11-11:
You need setting #define SERIAL_PORT_2 1, because Robin wifi interface usart is STM32 PA9,PA10 and it is Usart1

mks-viva, 2020-11-11:
@easycorexy 
You need setting #define SERIAL_PORT_2 1, because Robin wifi interface usart is STM32 PA9,PA10 and it is Usart1

mks-viva, 2020-11-20:
I will close this issues temporatity.

closed 2020-11-20 by mks-viva