makerbase-mks/MKS-WIFI #12: how to flash the modul via serial connection

RP-projekts, 2020-11-23:
After an error on setting the Modul to STA i need to flash it (used on an MKS TS35 with MKS Sgen_L V2.0). 
tried to flash the Modul via ch340 serial bridge (Arduino Uno, Arduino Mega) .. but i can't get sync via esptool used by Arduino/platformIO. same problem with spiffs 
i don`t have an tft35 board with special socket ---- just the harware mentioned above --- so i have to flash the new firmware via the Serial board (put VCC 5V, GND, Reset, RxD, Txd, and GPio0 to Low Voltage)

mks-viva, 2020-11-23:
Haha! You can refer to MKS SGEN_L V2 Wiki and i have a tutorial about how to use USB serial module flash ESP3d wifi firmware.
Link：https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_TFT_WIFI

RP-projekts, 2020-11-23:
Thanks for your prompt answer
i solved the problem using an CH340K (ws.cn) driver instead of the original ch340 
I ran into another bug: i can use the SD-Card .. i see all the files stored on the Card .. but connecting to the PC with the USB cable to MKS SGen_L board results with "device is not migrated error" -- is this an error with Win10 or are there any settings with Marlin 2.0 i can check?

mks-viva, 2020-11-24:
Can you provide a picture about the "device is not migrated error"?

RP-projekts, 2020-11-26:
sorry .. i was very bussy last days .. finally i got all the stuff working, except usb connection to any PC --but to tell the truth --- working with the mks wifi modul and using the mainboard (cause the bundle is sold with TS35 instead ofd tft35) is a pain in the neck -- i'm working on a new prototype of an high temperature fdm printer and i didn't expect so many problems with your board's
-- back to the hopefully last error -- connecting the mks sgen_l v2.0 via usb to an PC lasts in an error
[image]
[image]
[image]
looking forward to get the last problem solved
greetings from Germany

mks-viva, 2020-11-27:
I don't seem to understand what you mean. You want to use MKS SGEN_L V2's USB upload firmware?

DivingDuck, 2020-11-27:
He had the problem that the device isn't migrated in the OS. There is a mismatch in device drivers.
Maybe a good time to use a little helpful tool like [Zadig](https://zadig.akeo.ie/). It will help installing generic usb device drivers.

RP-projekts, 2020-11-27:
Thank's for your advice DivingDuck .. but it is not a problem with the USB Driver .. the same problem is with PCLinuxOs, FreeBSD and an Rasberry PI - i tried them all
I found some advice with the SKR-Board's -- Might be some Problem with TFT boards and PINS-- but no solution
I think "attach and release Media" isn't working proberly .. not sure it's a problem with Marlin or the hardware design
@mks-viva 
Uploading via onboard wifi (ESP3d) is working but taking hours, same with upload connecting Pronterface or Repetier
resolution : don't use MSK SGEN_L with the TS35 -- maybe with an TFT35 (or similiar) everything will work proberly
Question: can you fix the firmware?

RP-projekts, 2020-11-27:
// HAL idle task
void HAL_idletask() {
  #ifdef USE_USB_COMPOSITE
    #if HAS_SHARED_MEDIA
      // If Marlin is using the SD card we need to lock it to prevent access from
      // a PC via USB.
      // Other HALs use IS_SD_PRINTING() and IS_SD_FILE_OPEN() to check for access but
      // this will not reliably detect delete operations. To be safe we will lock
      // the disk if Marlin has it mounted. Unfortunately there is currently no way
      // to unmount the disk from the LCD menu.
      // if (IS_SD_PRINTING() || IS_SD_FILE_OPEN())
      /* copy from lpc1768 framework, should be fixed later for process HAS_SHARED_MEDIA*/
    #endif
    // process USB mass storage device class loop
    MarlinMSC.loop();
  #endif
}
-----------------------------
found this code above in hal.cpp ... maybe this is exactly the reason for the bug

mks-viva, 2020-11-28:
I think it’s a code problem, not related to hardware, very similar to this ISSUES：https://github.com/MarlinFirmware/Marlin/issues/19917

RP-projekts, 2020-11-28:
i'll follow MarlinFirmware/Marlin#19917 .. thank's for your prompt advice

closed 2020-11-28 by RP-projekts