makerbase-mks/MKS-WIFI #16: Board MKS Robin Nano V 1.x WiFi don't Work

Bighoneypot, 2021-01-16:
Good Morning, yestarday I have installed into my MKS Robin Nano V 1.x new Firmware version Marlin- bugfix2.0.x - MKS 2.1.2, but wifi don't work.
I have upgrade WiFi firmware with last version, but wifi don't Work https://github.com/makerbase-mks/MKS-WIFI

I have a question, about this problem.
There is a solution to fix this problem?
Also, can you hel me to downgrade firmware with a stable build??

I use Board MKS Robin Nano V 1.x with Ender 3 PRO.

[image]
[image]

Bighoneypot, 2021-01-16:
I have try this https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/tree/master/Nano%20firmware%20Marlin 

and for me work fine!!!

Bighoneypot, 2021-01-16:
I have try this https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/tree/master/Nano%20firmware%20Marlin 

and for me work fine.

But I need to mod source File

closed 2021-01-16 by Bighoneypot

RicPaiva, 2021-04-26:
Hi @Bighoneypot, 
I´m facing the same issue with Marlin- bugfix2.0.x - MKS 2.1.3.
You said you had to mod source file...
Do you have any idea how to solve it.
Thanks in advance...