makerbase-mks/MKS-WIFI #18: cannot build this code (Win10)

kralsam, 2021-03-06:
I cannot build this project. 

Console: 
```
MksHTTPUpdateServer.cpp:154:42: error: 'U_SPIFFS' was not declared in this scope
      res = Update.begin(3 * 1024 * 1024, U_SPIFFS);

.bla
.bla
.bla
'U_SPIFFS' was not declared in this scope
```
Arduino IDE V1.8.13
OS: Windows 10

Please Can you help me this issue?

pauloeduardosp, 2021-03-22:
Dear,
did you sove the problem, I'm try and get same error msg

kralsam, 2021-03-23:
No, didn't solved problem. So I think, I will coding new ESP8266 app.

BaltasarParreira, 2021-03-28:
To compile using the latest Arduino IDE (V1.8.13) and esp8266 core that comes with it and solve that error just change that line to:
 res = Update.begin(3 * 1024 * 1024, U_FLASH);

Them after that you going to have some other compile errors (3 if I remember) related to lines starting with:
 _currentClient.write...

on the "RepRapWebServer.cpp" file.

In all those lines with errors just delete the ", false" part in the end and leave the rest as is.
With this changes all code can be compiled and uploaded ok.

Looks like Makerbase are not using the latest IDE and ESP8266 core for creating the WIFI firmware.

Also... do you know that using a normal Nodemcu ESP8266 you can connect and adapt the pinout with some wiring work for using it as the WIFI module?
There is no need to buy the real WIFI module if you have one of this cheap boards laying around. ; )

kralsam, 2021-03-28:
Thanks for reply. Yes I know. I used generic nodemcu 12-F module. 
[image]

BaltasarParreira, 2021-03-28:
Ok, but that one programming directly from Arduino IDE can be done, but more complex on the connection with a NodeMCU like this just connect the USB straight to the PC. 
; )

https://opencircuit.shop/resources/content/14f8b53fd291a/crop/1200-630/NodeMcu-v3-ESP-12E-WIFI-Development-Board.jpg

kralsam, 2021-03-28:
I tried ESP12-F but it doesn't work, I think, it doesn't match MKS_robin_nanoV1.2 board.  It updated firmware this board from SD_card (MksWifi.bin) and ESP connected the wifi but doesn't work. (Original MKS_WIFI: ESP-12S)

BaltasarParreira, 2021-03-29:
How did you wired the ESP12-F to the MKS Robin Nano V1.2?

kralsam, 2021-03-29:
> How did you wired the ESP12-F to the MKS Robin Nano V1.2?

https://github.com/makerbase-mks/MKS-WIFI/blob/master/hardware/MKS%20Robin%20WIFI%20V1.1_001/MKS%20Robin%20WIFI%20V1.1_001%20SCH.pdf

BaltasarParreira, 2021-03-29:
For the ESP12-F module you need to add some pull up resistors in a few pins.
You can see those trough this schematic here:
https://github.com/makerbase-mks/MKS-WIFI/blob/master/hardware/MKS%20TFT%20WIFI%20V1.0_002/MKS%20TFT%20WIFI%20V1.0_002%20SCH.pdf

kralsam, 2021-03-29:
This parts on mks_robin_nano main board. Ref: Page: 8
 
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/blob/master/hardware/MKS%20Robin%20Nano%20V1.2_004/MKS%20Robin%20Nano%20V1.2_004%20SCH.pdf

BaltasarParreira, 2021-03-29:
Ok, never noted that the main board already had those parts.

But can you connect to the WIFI module from the Ultimaker Cura?
When you say is not working what's really the problem?

kralsam, 2021-03-29:
+ ESP firmware updated from SD card.
+ ESP creat AP SSID WiFi and configurable
+ ESP connect my local WiFi modem and request ip 
+ I see ESP MKS WIFI page on chrome with local ip address 

but 
- twotreesbluer WiFi screen not listed configuration page. 
- I can’t connect ESP with cura. 

I watch network wireshark. Cura send packages 8080 port but not reply mks board. (M codes in message packages)

BaltasarParreira, 2021-03-29:
Mine even I can connect and send files from cure not connected to the board.
You can test that first, just power the ESP12-F with the 3.3v and GND and see if Cura can connect.

BaltasarParreira, 2021-03-29:
I presume you have installed also the MKS WIFI Plugin for Cura and connected to the printer trough her IP on Cura?

kralsam, 2021-03-29:
> Mine even I can connect and send files from Cura not connected to the board.
> You can test that first, just power the ESP12-F with the 3.3v and GND and see if Cura can connect.

I will measure supply voltage when cura connecting

> Did you put the WIFI enable configuration and compile/upload the new firmware for the MKS Robin Nano?

Yes. Allready WiFi enabled board on printer. Also, I’m updated firmware latest(v2.0.x)

> I presume you have installed also the MKS WIFI Plugin for Cura and connected to the printer trough her IP on Cura?

Yes. Off course.

BaltasarParreira, 2021-03-29:
For doing the test as I said using only connection from Cura use this connections from the ESP12-F to the MKS Robin Nano board:
[image]

BaltasarParreira, 2021-03-29:
Since you already have the firmware on the ESP12-F, for using the rest of the functionalities you only need two more wires like this:

[image]

The other connections they exist only if you need to upload the firmware from the SD card again.

kralsam, 2021-03-29:
Can you try connection this wire connections ?

BaltasarParreira, 2021-03-29:
Yes, mine is working like that.
And as I said, only using the first connection example the printer doesn't really get files and see WIFI, but from Cura I can connect and send files.

kralsam, 2021-03-29:
Thanks for reply.
I understand you. But it not connected cura with MKS_WIFI.
Chrome Browser :
[image]

Cura : 
[image]

Can you try this mks robin nano v1.2 firmware please?
https://drive.google.com/file/d/1evR9P1amR1NoEbvMfiXvBZSnuyE71eiK/view?usp=sharing

BaltasarParreira, 2021-03-29:
I just compiled the WIFI firmware for your ESP-12-F.
I attached it here so you can install on your ESP, just use the normal procedure you used the first time to install it.

I don't have the printer now with me and is going to take time until I can do that test you asked, but later I'll show you my Cura connecting and sending files to my ESP even the not being connected to the printer board.

[MksWifi.zip](https://github.com/makerbase-mks/MKS-WIFI/files/6223655/MksWifi.zip)

kralsam, 2021-03-29:
I tried your binary but same result.

BaltasarParreira, 2021-03-29:
How are you registering on your WIFI network the module?
On that green page choose STA them fill up you SSID WIFI name and password so the module register, them try again.

kralsam, 2021-03-29:
> How are you registering on your WIFI network the module?
> On that green page choose STA them fill up you SSID WIFI name and password so the module register, them try again.

I retry update firmware of your send. It worked. I connected to cura. Thank you very much. Can you send me your source ?

BaltasarParreira, 2021-03-29:
Greate !!!
I attached my source here.
It is identical as the one here on github + the changes I mention so it compiles with the latest IDE and esp8266 core:
https://github.com/makerbase-mks/MKS-WIFI/tree/master/firmware_source/MksWifi
[Source_MksWifi.zip](https://github.com/makerbase-mks/MKS-WIFI/files/6224657/Source_MksWifi.zip)

You will need also the latest versions of all needed Libraries so before compiling do a Library update to see if they are all ok.

kralsam, 2021-03-29:
Thanks for all. 
Final result:
[image]

BaltasarParreira, 2021-03-29:
Yap, that's how it should work !

You are welcome.

closed 2021-03-31 by kralsam