makerbase-mks/MKS-WIFI #19: Random reboots on MKS Robin Nano 1.2 when WIFI module connected

BaltasarParreira, 2021-04-05:
Hi, I'm having issues lately after I compiled this source and installed on the module.
Any one facing same problems?

BaltasarParreira, 2021-04-08:
Update:
Just compile today the current Marlin 2.0 Bugfix Branch and the reboots disappear and it's stable.

Now the problem is printing from Cura using the MKS plugin the printer becomes like crazy moving all around here and there in all X, Y and Z but nothing really prints in the end.
So no normal operation.

saya812, 2021-05-03:
Hi! 
Same problem - printer reboots when Cura connect to wifi.
So if you start printer when cura is off - printer works. When you open cura on PC, cura write "printer connected" and printer reboots. If you start printer when cura is already opened on PC, printer reboos immediately.

foxlg, 2021-05-25:
Hello. I have the same problem as @saya812  (Cura 4.9.1 / 4.9.0). It is fine only with the official FW from the TwoTrees website, but their firmware has other problems and cannot be used :/