makerbase-mks/MKS-WIFI #23: How to configure platformio to compile ESP3D for this board ...

themacboy, 2021-05-04:
Hello,

I use Marlin Color UI, then Im forced to use ESP3D software (https://github.com/luc-github/ESP3D) to run my wi-fi.

But unfortunatelly Im not skilled on that and Im not sure how to configure platformio to run it.

Someone has do it before?

themacboy, 2021-05-05:
Hello? Is there any technician?

ColdBeer72, 2021-05-05:
Sorry, I'm not a technician (but we know each other from Telegram):
https://github.com/luc-github/ESP3D/issues/593

themacboy, 2021-05-06:
Hahaha I know that post hehehe but dint work for me.

After mkswifi.bin update, I load the Weui, then restart and when log again it says my firmarse is probably not correct :(

themacboy, 2021-05-06:
As I dont have any flasher hardware what I do is:

1. In VS code set env: esp8266dev
2. Compile and rename file to MksWifi.bin
3. First flash my MKS Robin Nano v.2 and v.3 with MKS UI (is the only way it loads the wifi firmware without a flasher)
4. Restart my printer and wait for the update.
5. Flash my MKS Robin Nano v.2 and v.3 with Marlin Color UI.
6. Connect to the ESP3D module from my mobile and load index.html.gz.
7. Restart all
8. Try to connect but receive a message that says my firmware is probably wrong hehehe

 I miss what Im doing worng !
 
 p.d: I know ! I must purchase a card flasher utility hahaha pending to see whats the correct one. ;)

ColdBeer72, 2021-05-06:
Hi again!

I made TWO changes on platformio.ini before compiling ESP3D:
```
default_envs = esp8266dev
board_build.ldscript = eagle.flash.4m3m.ld
```
Then, something like you with points 2, 3, 4 (I've got MKS_UI full time, so didn't need more changes). Step 6 made by me exactly like you. But I didn't need to restart anything: you upload index.html.gz and in the browser reload the page and configure it.

themacboy, 2021-05-06:
ummm with the MKS_UI what srial port dint you configured?