makerbase-mks/MKS-WIFI #24: About ESP3D ...

themacboy, 2021-05-06:
On the readmewhat means "the interface needs to be transferred to the AUX-1."

```
Compatibility with esp3d ?

Esp3d is a well known wifi project in 3d printing , too. If you are using MKS Robin series or MKS TFT series, mks wifi firmware can make the transfer speed up to 100KBytes/s. If you are using other 2560 series board, Esp3d firmware is more suitable. Esp3d firmware also can runs on MKS WIFI hardware, but, the interface needs to be transferred to the AUX-1.
```

AdamBear, 2021-05-19:
same question

iNDeX999, 2021-06-21:
same question too

BerranRemzi, 2021-12-09:
This tutorial was useful for me [Using_Web_Interface_or_SD_to_upload](https://www.mischianti.org/2021/11/28/mks-wifi-for-makerbase-robin-firmware-upgrade-and-new-web-socket-features-4/#Using_Web_Interface_or_SD_to_upload)

ImmortalTeknik, 2023-04-22:
Same question too...
In my case, loading the webui from the browser page is working but I still cant connect with the printer, but this is another topic.