makerbase-mks/MKS-WIFI #29: I release a custom version with Web Interface

xreef, 2021-11-29:
Hi,
this project is pretty frozen so i create a new project and add a new web interface.
I release It some days ago.
https://www.mischianti.org/category/project/web-interface-beeprint-for-mks-wifi/

I need help to do It better!!

[image]

[image]

Bye Renzo

BerranRemzi, 2021-12-09:
Perfect job

Stewie-040882, 2023-06-16:
Works Perfectly on the fokoos odin 5 f3.. but i Tried to compile my self to better understand and help develop but having no luck

xreef, 2023-06-17:
Hi,
I write a short guide to simplify the build process
https://mischianti.org/2021/11/21/mks-wifi-for-makerbase-robin-pcb-and-how-to-compile-upload-firmware-2/

Bye Renzo