makerbase-mks/MKS-WIFI #32: VIRUS within the needed ESP8266 core package you suggest to download

Majestix1970, 2022-10-26:
**WARNING!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!WARNING!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!**

**_ESPTOOL.EXE contained in the suggested download ESP8266 core package is infected with W32.Sality.AE VIRUS!!!_**
(according your advice to download the package: Please directly download the [esp8266 core on MKS Github](https://github.com/makerbase-mks/Esp8266-Core-For-Arduino) to the arduino data path, eg, "C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266".)
Also reported by different other users on https://github.com/makerbase-mks/Esp8266-Core-For-Arduino (see issues which are open for a long time already!!!)
Can youe please ensure that the infected ESP8266 core package gets removed and replaced with a clean one please!

MKS WiFi also has "strange" compile issues (may be because of the ESP8266 core issue....)
can you please advice how to compile 
THANK YOU

NotReallyADeveloper, 2023-03-01:
Was this ever acknowledged/resolved?

Majestix1970, 2023-03-02:
unfortunately not to my knowledge!!!